KiCad/kicad-symbols #1957: 5.1.3 library release

evanshultz, 2019-07-08:
Hey all! 5.1.3 is near. See Wayne's message at https://lists.launchpad.net/kicad-developers/msg41442.html.

What in the library needs to be done? For issues/PRs with a 5.1.3 milestone I only find https://github.com/KiCad/kicad-footprints/issues?q=is%3Aopen+is%3Aissue+milestone%3A5.1.3 and https://github.com/KiCad/kicad-packages3D/pull/565.

Here are issues and PRs with a "bug" label:
- Symbol issue: https://github.com/KiCad/kicad-symbols/pull/1649
- Symbol PR: https://github.com/KiCad/kicad-symbols/issues?q=is%3Aopen+is%3Aissue+label%3ABug (8)
- Footprint issue: https://github.com/KiCad/kicad-footprints/issues?q=is%3Aopen+is%3Aissue+label%3ABug
- Footprint PR: https://github.com/KiCad/kicad-footprints/pulls?q=is%3Aopen+is%3Apr+label%3ABug
- 3D issue: https://github.com/KiCad/kicad-packages3D/issues?q=is%3Aopen+is%3Aissue+label%3ABug
- 3D PR: https://github.com/KiCad/kicad-packages3D/pulls?q=is%3Aopen+is%3Apr+label%3ABug

Does anybody have the time and initiate to sort out any of these?

antoniovazquezblanco, 2019-07-09:
Went for some easy ones:
https://github.com/KiCad/kicad-packages3D/pull/565
https://github.com/KiCad/kicad-packages3D-source/pull/228
https://github.com/KiCad/kicad-packages3D/pull/284

Added requested changes in https://github.com/KiCad/kicad-footprints/issues/1663 to existing pr https://github.com/KiCad/kicad-footprints/pull/1670 in order to close both if you want to review them @evanshultz 

Thanks!

antoniovazquezblanco, 2019-07-10:
Also have taken on the only bug issue in the 3D repo:
https://github.com/KiCad/kicad-packages3D/issues/546

https://github.com/KiCad/kicad-packages3D/pull/593

poeschlr, 2019-07-23:
It seems the release is imminent. So i will tag as soon as i tested the symbol lib for incompatibilites.

poeschlr, 2019-07-23:
Wow there has basically been not much done on the symbol side since 5.1.2. (Similar story with footprints)
We might need some more maintainers to be honest.

The only dangerous changes found:
Pins have been moved, renumbered or removed in symbol:
- 'Driver_LED.lib:IS31FL3216'
- 'Logic_LevelTranslator.lib:TXB0108DQSR'
- 'Potentiometer_Digital.lib:DS1267_SOIC'
- 'Sensor_Optical.lib:BPW21'

Normal pins ok but NC pins have been moved, renumbered or removed in symbol: 
- 'Interface_CAN_LIN.lib:TJA1029T'
- 'Interface_CAN_LIN.lib:TJA1029TK'
- 'Sensor_Pressure.lib:40PC015G'
- 'Sensor_Pressure.lib:MPXA6115A'

antoniovazquezblanco, 2019-07-24:
I think we were more cautious I believe. The rules are clearer for me now. That doesn't mean we need more maintainters, I agree.

Thanks for your effort!

closed 2019-07-24 by antoniovazquezblanco

poeschlr, 2019-08-04:
The 5.1.3 release had a annoying bug so it was decided to go directly to 5.1.4 instead. To speed up the release i tagged the same commit.