KiCad/kicad-symbols #1969: Asymmetric TVS diode array

chschlue, 2019-07-12:
I am new to KiCad (coming from Eagle) and would like to contribute some symbols.

I need [this asymmetric TVS array](https://m.littelfuse.com/~/media/electronics/datasheets/tvs_diode_arrays/littelfuse_tvs_diode_array_sm712_datasheet.pdf.pdf) for a design.
Should I create a new symbol for this specific part or just use the generic D_TVS_x2_* symbols?

poeschlr, 2019-07-12:
In this case i would suggest the creation of a fully specified part just because there are no asymmetric symbols in the lib right now. I would advice against adding generic symbols for asymmetric parts as one might want to include the voltage values in the symbol. This is not really possible in a generic way.

---

In general we can not give advice in this question. We do not know your preferences and can therefore not really make the decision for you. Both the fully specified and generic workflow have their place. (I personally prefer fully specified but others might prefer the generic workflow.)

chschlue, 2019-07-12:
PR #1973

closed 2019-07-12 by chschlue