KiCad/kicad-symbols #2115: Distinguishing open-collector from push-pull for isolators

matthijskooijman, 2019-08-30:
I'm working on a symbol for an optocoupler, and I'm a bit in doubt how to proceed wrt distinguishing between open-collector and push-pull images..

I'm working on the Si8735, which is drawn in the datasheet as follows:
[image]

 It's quite a complex thing, but I've managed to pretty much replicate it in a symbol:

[image]

This is somewhat inspired by the symbol for the HCPL-261A (I also tried to make it a drop-in replacement symbol, which is why my Si8735 is a bit cramped). The HCLP-261A symbol is fairly explicit about what is inside (a buffer with an NPN to indicate open-collector).

[image]

However, I'm worried that this does not really scale well (for dual or quad designs, it gets messy quickly). Looking around, I found that some optos (both in kicad and in datasheets) use an "and" symbol or a "buffer" symbol on the output, which can be a lot more compact.

[image]

However, it suggests to me that the output is push-pull, while they often are open-collector. I found the HCPL-2202 (datasheet image shown above), which is push-pull which uses an "and" symbol with up/down arrows and thought that that might be an indication of push-pull output, but I can't find any reference to that and I suspect it might mean Schmitt Trigger instead, though that is typically drawn without arrows. The TLP2745 datasheet also suggests that the arrow up/down symbol is indeed Schmitt Trigger, and does not indicate push-pull:

[image]

Looking for other examples of push/pull vs OC in the kicad library, I found the symbols for the 74LS00 (push-pull NAND) and 74LS01 (open collector NAND), which simply look identical as well...

 
[image]

Is there any existing convention to consisely distinguish these (without VDD or GND wires to keep things clean)? Any other suggestions on how to draw these?

Also, it seems that there are also some different drawing styles used in optoisolators (at least in the logic isolator department), as already shown in some of the images above. I suspect that they are mostly based on datasheet images, but having a bit more consistency in the Kicad Isolator library might be a good idea regardless?

@Misca1234 & @jkriege2, you were involved in a recent isolator-library cleanup (#287), maybe you have some input?

evanshultz, 2019-08-30:
First, where internal wire cross, place a 10mil filled circle to indicate connectivity.

Yes, things get cramped and don't scale well. That's true. I guess I would fall on the side of keeping things small where possible instead of making smaller/simpler symbols bigger than required. That all being said, the logic library (and logic symbols used elsewhere, like in optos), is due for an overhaul. Hopefully for v6.

I like simpler symbols instead of drawing out the discrete transistors. And I'm sure that it's a Schmitt trigger symbol, although the arrows aren't familiar to me either.

See page 11 of http://www.ti.com/lit/ml/sdyz001a/sdyz001a.pdf. I don't have the IEEE doc, but if this is correct the appropriate symbol to indicate open collector output is a diamond with a line under it. I haven't seen this done in practice, by the way, but I noted this at one point and thought it might be useful in the future.

Note sure if this is helpful but hopefully it adds positively to your issue.

calebreister, 2019-08-30:
@evanshultz The "⎐" symbol to denote an open-collector output is common in the United States (apparently common enough to have a Unicode code point). According to [Wikipedia](https://en.wikipedia.org/wiki/Open_collector), there are multiple variations to denote different corner cases, though I've only seen "⎐" used in practice.

matthijskooijman, 2019-08-31:
> First, where internal wire cross, place a 10mil filled circle to indicate connectivity.

Do you also mean on T-junctions? Since that does not seem to be done in practice. For crossing junctions, that seems like a good idea, but I do not think there are any in my images above?

> the appropriate symbol to indicate open collector output is a diamond with a line under it.

Ah, nice. If we want to add something, it would be a good to adhere to a standard, so this might be useful. It would be even better if the symbol is also used in practice (so people will recognize it right away), but when used consistently within the library, people will probably learn the meaning after a while. Also, good to hear from @calebreister that he has seen this symbol in common practice as well.

Also having an explicit symbol for push-pull would be even nicer, but it seems this is just the default and has no symbol.

So, to me it would be good to use this symbol for OC outputs on logic symbols everywhere. To have something for further discussion, there's a proposal for a KLC addition about this below. Note that this only concerns outputs with graphical symbols, for regular pin outputs, pin functions must be clear from the pin names, or the datasheet, so no need to add symbols there.

How about adding this to [KLC S3.4](http://kicad-pcb.org/libraries/klc/S3.4/):

> When logic symbols (such as AND ports or buffers) are used, these are assumed to have push-pull outputs. When the outputs are open-collector, open-drain, open-emitter or open-source instead, this must be noted by the corresponding IEEE91-1984 symbol:
> - ⎐ for a pin that outputs low-Z L or hi-Z H (or ⎒ with an internal pull-up resistor)
> - ⎏ for a pin that outputs hi-Z L or low-Z H (or ⎑ with an internal pull-down resistor)
>
> Example push-pull vs open-collector:
[image]

Alternatively, we could put the symbol on the outside of the logic symbol, to keep room inside for e.g. Schmitt Trigger symbols, or perhaps put them side-by-side (I do not think the TLP2704 actualy has Schmitt Trigger, but just to illustrate the point).

[image]

I also just realized that the above are not completly correct examples yet: To make these diagrams as simple as possible (and allow scaling up), the VDD and GND wires should not be drawn (and maybe the pins for them should be at the right as well, to emphasize they are on the right side of the isolation barrier). I've also been messing around with the line width, but that can be fixed later.

For completeness, I've included all four versions of the symbol in the text (open collector/emitter with/without pullup), though I haven't seen open-emitter much in practice (internal pullup is somewhat more common at least in Isolators I think, but might also be too much detail - open-collector with pullup could be without symbol since it behaves similar to push-pull). So maybe just the open-collector symbol is sufficient?

calebreister, 2019-08-31:
@matthijskooijman I've seen both the 2nd and 3rd options used for open collector pins. However, I am questioning whether it is necessary to indicate an open collector output on the symbol due to the fact that the pin type should already be specified.

Also, I would recommend simplifying the Schmitt trigger symbol by removing the arrows. I haven't seen this used much in practice, and it adds some clutter.

matthijskooijman, 2019-09-01:
> However, I am questioning whether it is necessary to indicate an open collector output on the symbol due to the fact that the pin type should already be specified.

The output type is used for DRC, but not normally shown in schematics. The usecase I'm trying to improve here is reading a schematic that uses graphical diagrams. Another approach could be to somehow let KiCad mark the difference between different output types in an automated manner, but that would require changes in KiCad and it might be tricky to find an automatic annotation that looks good in all cases.

> Also, I would recommend simplifying the Schmitt trigger symbol by removing the arrows. I haven't seen this used much in practice, and it adds some clutter.

Probably a good idea, I just copied them from an existing symbol like this (TLP2745 and friends IIRC).

chschlue, 2019-09-27:
Just for the record and because this came up again in #2087: 
I'm also against the arrows. No need for an excessively pedantic textbook hysteresis plot here.

hildogjr, 2020-04-08:
A implementation to better behavior, keeping some symbols always up-righted,  a discussion for implementation had start at https://gitlab.com/kicad/code/kicad/-/issues/4049