KiCad/kicad-symbols #2135: Switching regulator switch pin type

chschlue, 2019-09-07:
Partially related to issue #1711 

What's the preferred electrical type for switch pins of switching regulators such as https://www.analog.com/media/en/technical-documentation/data-sheets/ADP1614.pdf?

Existing symbols use nearly everything: power output, output, open collector, passive, I think I've even seen input.
Most of these seem technically correct (especially power output and open collector), except not really.

evanshultz, 2019-09-07:
I have discussed this in several PRs and also started an issue with these (non-digital) pins at https://github.com/KiCad/kicad-symbols/issues/381.

This isn't part of KLC so I will merge if it makes sense, but usually ask for Passive.

Why? The ERC engine in KiCad is meant to check for 'digital' or 'logic' pins and these don't apply. But this pin isn't connected to the net which actually supplies power. Buck circuits have an inductor after the switch pin, boost has a diode, and a flyback has a transformer and then a diode. There are other circuit types that IC regulators can perform (Cuk, inverter, etc.) but all that pop into my head also match.

Therefore, if the switch pin is Power Output it can supply other Power Input pins without complaints in KiCad. And that can't be correct because it would indicate a part is missing in the schematic. So Power Output seems like definitely the wrong type.

They can safely be connected to a variety of things in odd or unique circuits, such as possibly being used for synchonization or a switching detector. But in those cases, I imagine a passive part being connected to the IC's switch pin so the pin type doesn't matter too much.

Output wouldn't be so bad, but since KiCad's ERC engine doesn't really capture the function of this pin type Passive seems like the most safe choice to me.

Also see https://github.com/KiCad/kicad-website/issues/342, where Rene listed a hierarchy of pin types. This should migrate to KLC but seems to have been missed.

poeschlr, 2019-09-08:
My issue on the website simply never got codified as i had (and still have) other klc pull requests open. I try to avoid working on too many things in parallel.

closed 2019-09-13 by chschlue