KiCad/kicad-symbols #2142: 5.1.5 is planned for sometime mid september

poeschlr, 2019-09-10:
See https://lists.launchpad.net/kicad-developers/msg41959.html

please add pull requests that you think should go into 5.1.5 such that librarians know what to prioritize.

I personally would like to see https://github.com/KiCad/kicad-footprints/pull/1821 in 5.1.5

evanshultz, 2019-09-10:
I didn't see Wayne agree to Seth's proposed schedule, but with those bugs I assume it won't be long.

- I am going to wrap up the footprints in https://github.com/KiCad/kicad-footprints/issues/608.
- The PR you noted above is also something I can help with.
- While I didn't look at Issues and PRs marked as bugs, I know of https://github.com/KiCad/kicad-symbols/pull/2141 and https://github.com/KiCad/kicad-footprints/pull/1835 that I would like to see included.
- I intend to get finally get https://github.com/KiCad/kicad-templates/pull/25 merged.

poeschlr, 2019-10-21:
Ok it seems like 5.1.5 is now getting some traction: https://lists.launchpad.net/kicad-developers/msg42278.html

antoniovazquezblanco, 2019-10-21:
It seems that there is no urgent symbol and 3D stuff; footprints are only missing the rounding PRs and the ISO templates that Evan mentioned avobe:

https://github.com/KiCad/kicad-symbols/milestone/10
https://github.com/KiCad/kicad-footprints/milestone/6
https://github.com/KiCad/kicad-packages3D/milestone/5
https://github.com/KiCad/kicad-templates/milestone/4

poeschlr, 2019-11-17:
The source code has been tagged so i will tag the library within the next hour or so.

poeschlr, 2019-11-17:
Report regarding symbol lib:

Pins have been moved, renumbered or removed in symbol
- '74xx_IEEE.lib:74278' (#2162)
- '74xx_IEEE.lib:74LS192' (#2162)
- '74xx_IEEE.lib:74LS193' alias of 74LS192 (#2162)
- '74xx_IEEE.lib:74LS290' (#2162)
- '74xx_IEEE.lib:74LS49' (#2162)
- 'Analog.lib:LF398_DIP8' (#2141) -> Pin number fix
- 'Analog.lib:LF398_SOIC8' alias of LF398_DIP8 (#2141) -> Pin number fix
- 'Analog.lib:LF398_SOIC14' (#2141) -> Pin number fix
- 'CPU.lib:P4080-BGA1295' (#2239) -> right side pins where positioned wrong with regard to outline
- 'Diode.lib:BAV99' (#2131) -> pin was duplicated over multiple units
- 'MCU_Microchip_SAME.lib:ATSAME54N19A-A' (#2270) -> pins have been moved
- 'Sensor_Humidity.lib:HDC1080' (#2030) -> only pin rename

The changes to the 74xx_IEEE lib are mixed in with file version update so it is hard to tell what changed. According to pull request description the pin names have been fixed.

---

Removed:
- 'Amplifier_Operational.lib:MCP6561U' was an alias of LMV321
- 'Amplifier_Operational.lib:MCP6561x-LT' was an alias of MCP6001x-LT
- 'Amplifier_Operational.lib:MCP6561R' was an alias of MCP6L91RT-EOT
- 'Amplifier_Operational.lib:MCP6561-OT' was an alias of MCP6L91T-EOT

---

Normal pins ok but NC pins have been moved, renumbered or removed in symbol 
- 'Connector.lib:AVR-ISP-10'
- 'Connector.lib:AVR-PDI-6'
- 'Connector.lib:AVR-TPI-6'
- 'MCU_Module.lib:Arduino_Leonardo'
- 'MCU_Module.lib:Arduino_UNO_R2'
- 'MCU_Module.lib:Arduino_UNO_R3'
- 'Regulator_Linear.lib:TLV70012_SOT23-5'
- 'Regulator_Linear.lib:TLV70013_SOT23-5'
- 'Regulator_Linear.lib:TLV70015_SOT23-5'
- 'Regulator_Linear.lib:TLV70018_SOT23-5'
- 'Regulator_Linear.lib:TLV70019_SOT23-5'
- 'Regulator_Linear.lib:TLV70025_SOT23-5'
- 'Regulator_Linear.lib:TLV70030_SOT23-5'
- 'Regulator_Linear.lib:TLV70032_SOT23-5'
- 'Regulator_Linear.lib:TLV70033_SOT23-5'
- 'Regulator_Linear.lib:TLV70036_SOT23-5'
- 'Regulator_Linear.lib:TLV70212_SOT23-5'
- 'Regulator_Linear.lib:TLV70215_SOT23-5'
- 'Regulator_Linear.lib:TLV70218_SOT23-5'
- 'Regulator_Linear.lib:TLV70225_SOT23-5'
- 'Regulator_Linear.lib:TLV70228_SOT23-5'
- 'Regulator_Linear.lib:TLV70230_SOT23-5'
- 'Regulator_Linear.lib:TLV70231_SOT23-5'
- 'Regulator_Linear.lib:TLV70233_SOT23-5'
- 'Regulator_Linear.lib:TLV70235_SOT23-5'
- 'Regulator_Linear.lib:TLV70237_SOT23-5'
- 'Regulator_Linear.lib:TLV70245_SOT23-5'
- 'Regulator_Linear.lib:TLV71209_SOT23-5'
- 'Regulator_Linear.lib:TLV71210_SOT23-5'
- 'Regulator_Linear.lib:TLV71211_SOT23-5'
- 'Regulator_Linear.lib:TLV71310PDBV' alias
- 'Regulator_Linear.lib:TLV71311PDBV' alias
- 'Regulator_Linear.lib:TLV71312PDBV' alias
- 'Regulator_Linear.lib:TLV71315PDBV' alias
- 'Regulator_Linear.lib:TLV71318PDBV' alias
- 'Regulator_Linear.lib:TLV713185PDBV' alias
- 'Regulator_Linear.lib:TLV71325PDBV' alias
- 'Regulator_Linear.lib:TLV71328PDBV' alias
- 'Regulator_Linear.lib:TLV713285PDBV' alias
- 'Regulator_Linear.lib:TLV71330PDBV' alias
- 'Regulator_Linear.lib:TLV71333PDBV' alias
- 'Regulator_Linear.lib:TLV75509PDBV' alias
- 'Regulator_Linear.lib:TLV75510PDBV' alias
- 'Regulator_Linear.lib:TLV75512PDBV' alias
- 'Regulator_Linear.lib:TLV75515PDBV' alias
- 'Regulator_Linear.lib:TLV75518PDBV' alias
- 'Regulator_Linear.lib:TLV75519PDBV' alias
- 'Regulator_Linear.lib:TLV75525PDBV' alias
- 'Regulator_Linear.lib:TLV75528PDBV' alias
- 'Regulator_Linear.lib:TLV75529PDBV' alias
- 'Regulator_Linear.lib:TLV75530PDBV' alias
- 'Regulator_Linear.lib:TLV75533PDBV' alias
- 'Regulator_Linear.lib:TLV702475_SOT23-5'
- 'Relay_SolidState.lib:FOD420'
- 'Relay_SolidState.lib:FOD4208' alias of
- 'Relay_SolidState.lib:FOD4216' alias of
- 'Relay_SolidState.lib:FOD4218' alias of
- 'Relay_SolidState.lib:MOC3010M'
- 'Relay_SolidState.lib:MOC3011M' alias of
- 'Relay_SolidState.lib:MOC3012M' alias of
- 'Relay_SolidState.lib:MOC3020M' alias of
- 'Relay_SolidState.lib:MOC3021M' alias of
- 'Relay_SolidState.lib:MOC3022M' alias of
- 'Relay_SolidState.lib:MOC3023M' alias of
- 'Relay_SolidState.lib:MOC3051M' alias of
- 'Relay_SolidState.lib:MOC3052M' alias of
- 'Relay_SolidState.lib:MOC3031M'
- 'Relay_SolidState.lib:MOC3032M' alias of
- 'Relay_SolidState.lib:MOC3033M' alias of
- 'Relay_SolidState.lib:MOC3041M' alias of
- 'Relay_SolidState.lib:MOC3042M' alias of
- 'Relay_SolidState.lib:MOC3043M' alias of
- 'Relay_SolidState.lib:MOC3081M' alias of
- 'Relay_SolidState.lib:MOC3082M' alias of
- 'Relay_SolidState.lib:MOC3083M' alias of
- 'Relay_SolidState.lib:MOC3061M' alias of
- 'Relay_SolidState.lib:MOC3062M' alias of
- 'Relay_SolidState.lib:MOC3063M' alias of
- 'Relay_SolidState.lib:MOC3162M' alias of
- 'Relay_SolidState.lib:MOC3163M' alias of
- 'RF.lib:ADP-2-1W'
- 'RF.lib:LRPS-2-1'

closed 2019-11-17 by poeschlr