KiCad/kicad-symbols #2145: Commit own symbols / footprint / packages

silelis, 2019-09-11:
Hello,

I am just starting with KiCAD (switching from Eagle2KiCAD) but have some amount of symbols / footprint / packages like bleutooth, FM breakouts, mp3 schips and so on.

My question is how and if I can share it with this git? I don't want to loose theme and can share with community

antoniovazquezblanco, 2019-09-11:
Yes! Your contributions are very welcomed!

You have brief instructions in http://kicad-pcb.org/libraries/contribute/ on how to proceed.

Thank you!

evanshultz, 2019-09-11:
@silelis 
Welcome!

Note that there are guidelines for this library at http://kicad-pcb.org/libraries/klc/ which contributions need to adhere to. Your symbols/footprints may need updating before they can be included in the library here.

Since I hope your question is answered please close this issue when you have a chance. Or if not, add another comment.

DanSGiesbrecht, 2019-11-02:
Unless anyone has an objection, I'm going to close this.

closed 2019-11-02 by DanSGiesbrecht