KiCad/kicad-symbols #2163: PZTA42 has 3 pins in symbol, 4 pins in footprint

gkeeth, 2019-09-14:
PZTA42 is a SOT-223 NPN BJT ([datasheet](https://www.onsemi.com/pub/Collateral/PZTA42T1-D.PDF))

The pin layout of the physical part is BCEC, i.e. 1-Base, 2,4-Collector, 3-Emitter. See screenshot from datasheet:

[image]

 

The current Kicad symbol is basically an NPN-BCE, i.e. 1-Base, 2-Collector, 3-Emitter. 
The symbol assigns SOT-223 as the default footprint.

[image][image]

The problem is SOT-223 is a 4-pin footprint, but the symbol has only 3 pins. Pin 4 in the footprint should correspond to the collector.

I can think of two solutions:
1. change the PZTA42 symbol to be essentially Q_NPN_BCEC, which would match the existing SOT-223 footprint
2. Keep the PZTA42 symbol as is, but change the default footprint to SOT-223-3_TabPin2. This will effectively connect pins 2 and 4 in the footprint rather than the schematic. 

I think 1 is the better solution because then the kicad symbol will match the datasheet. It is a symbol pin change, which will break any schematics currently using this symbol, but is this a big problem if it's already broken?

2 has the advantage of not breaking existing schematics (I think) but then the symbol doesn't match the datasheet, and it will use a non-standard footprint for no real reason. 

I'm happy to submit a pull request if I can get some guidance on which solution to go for.

gkeeth, 2019-09-15:
I dug around a little more, and this affects all SOT-223 transistors in the library. 
These parts also have 3 pins in the symbol but specify the standard (4 pin) SOT-223 footprint:
BCP51
PZT2222A
PZT3904 (alias BCP56)
PZT3906 (alias BCP53, PBSS301PZ)
PZTA42

evanshultz, 2019-09-16:
Option 2 matches the convention used in the library, and won't break existing schematics, so let's go with that.

However, I've noted before that this is a limitation because it may be valid to only connect pin 2 or only pin 4 (in the footprint shown above) if thermal isn't a problem. In that case, the default symbol doesn't work. So switching to a 4-pin symbol for v6 is, in my opinion, something to really consider.

gkeeth, 2019-09-16:
Out of curiosity, can you elaborate on the convention used in the library (or just point me to another illustrative part)?

When I get a minute I'll put together a quick pull request to change the footprints and link it here.

evanshultz, 2019-09-16:
It's just the convention used, and not documented. Two related discussions not involving me and going back some time to establish the precedent are https://github.com/KiCad/kicad-library/pull/1875 and https://github.com/KiCad/kicad-library/pull/479. (Note that whether using the footprint with two pin 2s or stacking two symbol pins with the footprint shown above, the result is the same to the user in that only three schematic pins can be assigned.)

Other footprint families which have a similar paradigm are TO252 and TO263.

gkeeth, 2019-09-17:
I created #2175 to fix this.

evanshultz, 2019-09-17:
Removed the bug label since the fixing PR has been merged. Marked with v6 milestone as this is something we can revisit at that time.