KiCad/kicad-symbols #2178: XPT2046TS footprint is incorrect

pyrohaz, 2019-09-17:
Footprint is "Package_SO:TSSOP-14_4.4x5mm_P0.65mm" but should be "Package_SO:TSSOP-16_4.4x5mm_P0.65mm"

evanshultz, 2019-09-17:
Do you have an official datasheet link? What you say is confirmed by https://www.buydisplay.com/download/ic/XPT2046.pdf, but I'm not able to view XPTEK's website.

pyrohaz, 2019-09-17:
Yeh, the Xptek website doesn't seem to have it on but LCSC has the same datasheet as featured above: https://lcsc.com/product-detail/Driver-High-Current_XPT2046_C19076.html I also have a couple of the ICs here too.

poeschlr, 2019-09-18:
The symbol defines pin 15 and pin 16 so yes there is some mistake. (Either too many pins in the symbol or the wrong footprint. And as datasheets seem to suggest the later then i would guess this to be the correct fix.)

evanshultz, 2019-09-19:
Without knowing the body size and pitch this is probably the best we can do.

@pyrohaz 
Do you have one of the ICs that you can confirm the footprint? Also, are you willing to make a PR to fix this error?

chschlue, 2019-09-19:
FYI:
XPTEK's website loads for me now, but their English DS links are broken.

evanshultz, 2019-09-20:
http://www.xptek.cn/uploadfile/download/201707171401161883.pdf is the Chinese datasheet and it has dimensions. I submitted https://github.com/KiCad/kicad-symbols/pull/2183.

closed 2019-09-21 by diegoherranz