KiCad/kicad-symbols #2184: How to name symbols for 'equivalent' parts with different MPNs

evanshultz, 2019-09-20:
Picking off a conversation from . I believe the idea was originated at .

evanshultz, 2019-09-20:
Edit: Someone pressed a keyboard key and submitted this early. Oops.

poeschlr, 2019-09-20:
To clarify this is about a subsection of S2.1 that was touched in https://github.com/KiCad/kicad-website/pull/407

The wording before the KLC change:
- Part name should include extension for specific footprint if the same part is produced by multiple manufacturers that have conflicting package name suffixes (e.g. `SOIC`)
My original suggested wording:
- If and only if multiple manufacturers build compatible devices then the common base part number is used with clear name suffix for the footprint (e.g. `L78L05_TO92`)

Wording suggested by evan:
- If multiple manufacturers produce compatible parts but their part number suffixes for specifying the package differ, then the common base part number is used with clear name suffix for the footprint (e.g. `L78L05_TO92`)

---

The problem:
It seems the example listed (which was one of the original reasons why this rule even exists) would violate the rule by evan (and probably also the original wording)

poeschlr, 2019-09-20:
And maybe stating the obvious: If the current wording is kept (yes it got merged) then we have quite a few symbols to rename for v6.

And if we already go this far why not get rid of this exception as it only creates more work for us and is not really well liked anyways.

evanshultz, 2019-09-20:
I think we're on the same page that this is hacky and it would be nice if we found a better way. I noted a fully-defined example above where we could use an ALIAS to capture two identical PNs. Kind annoying to have to maintain two identical DCM entries, but this doesn't happen often and fully-specified may be the best method for KiCad library users. Food for thought...

chschlue, 2019-09-20:
The discussion from https://github.com/KiCad/kicad-website/pull/407 has shown (I think) that narrowing the exception down to just suffixes makes it basically useless because it would only apply to a handful of parts (e.g. 6N137). Broadening it to "base MPNs" makes it wishy-washy because it is not always clear what the easily recognizable "base MPN" is supposed to be when prefixes and possibly more differ.
It seems to me that going with aliases and actual package suffixes makes everything cleaner and clearer.

Some examples:

Current symbol: L78L05_TO92 (ST prefix but covers ON, TI and many more)
Alternative: L78L05xxZ (ST), alias MC78L05xxP (ON), alias UA78L05xxLP (TI), more as required

Current symbol: 6N137 (no suffix, alias of HCPL-261A, specified for DIP-8)
Alternative: 6N137 (Vishay, Broadcom), alias 6N137xM (ON)

Current symbol: 74HC04 (generic)
Alternative (SOIC-14): 74HC04D (Nexperia), alias MC74HC04AD (ON), alias SN74HC04D (TI)

Pro:
Rule is always clear

Con:
Package suffixes might conflict
A lot of work

poeschlr, 2019-09-21:
The logic libs (your 74HC04 example) do not have footprints assigned so they are considered generic symbols and are not covered by this rule for this reason.

We can make the fully specified but well lets say i am not optimistic that we will find a volunteer for this thankless task (we could not even find enough manpower for fixing the hidden power pin problem)

chschlue, 2019-09-21:
I know it's probably not going to happen soon, but all the parts covered by S2.1.3 and by my examples are basically generic (in that they have several manufactures using the same "base MPN"), so my suggestions fewer rules and, more importantly, fewer exceptions.
Or is there a particular reason for keeping generic libs at all, except for the lack of manpower?

poeschlr, 2019-09-21:
We will always have some generic libs where it makes sense. Connectors for example make no sense to be made fully specified. Same with standard passives.

chschlue, 2019-09-21:
I thought the context implied we're talking about ICs here.
And in that realm, I fail to see the real difference between 4000 series, 7400 series, 78/79xx regulators, "generic" op amps and so on.