KiCad/kicad-symbols #2200: BFR92 has wrong package assigned

evanshultz, 2019-09-24:
This is concerning Transistor_BJT.lib as I was making https://github.com/KiCad/kicad-symbols/pull/2199.

Our `BFR92` symbol is assigned the SOT-323 package. It references a NXP datasheet at a broken URL. I don't find BFR92 with no suffix available on the market. So I'm not sure what's going on.

Sticking with NXP datashseets, I do see BFR92AW at https://www.nxp.com/docs/en/data-sheet/BFR92AW_N.pdf? in SOT-323. That is apparently the part intended to be created?

Question: Do we consider this a bug and are free to rename the symbol to `BFR92AW`?

diegoherranz, 2019-10-14:
I can't find anything about the BFR92 without suffixes either. Only BFR92AW (SOT-323) or BFR92A (SOT-23).

Even if this breaking change would be an easy one to fix by the user in the schematic, I think we need to avoid breaking symbols unless it is a bug. I'm not sure I would consider this a bug.

I would vote for leaving this for v6, when we can rename this to BFR92AW and probably create the BFR92A at the same time to make it clear. Either way, they could be aliases, I think, and that doesn't need to wait for v6.

Just after writing this I thought, what if we add "BFR92AW" in the description or keywords until this can finally get fixed properly?

Thanks!

evanshultz, 2019-10-15:
Thanks for looking and responding!

I personally lean toward this being a bug but with low impact so waiting or not are OK to me.

Because the package is different they can be ALIASes. Or did I misunderstand?

Add the keyword now is a good idea. Updating the datasheet would also be good.

diegoherranz, 2019-10-15:
> I personally lean toward this being a bug but with low impact so waiting or not are OK to me.

I wouldn't be opposed to that if other librarians agree with it.

> Because the package is different they can be ALIASes. Or did I misunderstand?

What I meant is that BFR92 could be an alias of BC817W, for instance, instead of having its own symbol.

klepklep17, 2020-05-30:
Many times NXP uses extension as an other version of the packages. The BFR92 is technologically probably the same as the BFR92AW only in an other package.

You could refer to the BFR92 version of Vishay. It seems the almost the same.

[VISHS63618-1.pdf](https://github.com/KiCad/kicad-symbols/files/4706176/VISHS63618-1.pdf)
[BFR92_PhilipsSemiconductors.pdf](https://github.com/KiCad/kicad-symbols/files/4706184/BFR92_PhilipsSemiconductors.pdf)