KiCad/kicad-symbols #2208: LP2950 wrong pins

Enzo1960, 2019-09-26:
LP2950 voltage regulator has input and output terminals exchanged. Furthermore, if I try to modify it or any other of LP2950 family, the editor bring me at MC78L05.

chschlue, 2019-09-26:
What do you mean?
LP2950-*_TO252 are aliases of LM78M05_TO252, pinout 1-IN, 2-GND, 3-OUT.

evanshultz, 2019-09-26:
Here is the symbol:
[image]

And the default footprint:
[image]

A quick Digikey search shows two LP2950 vendors in TO252 package:
- http://www.ti.com/lit/ds/symlink/lp2951-n.pdf
- https://www.onsemi.cn/PowerSolutions/document/LP2950-D.PDF

The pinouts match. So I agree with @chschlue that the problem is not obvious. Can you clarify what you found?

Regarding the MC78L05 thing, that is the 'root' part which has many ALIASes. So if you click on an ALIAS, the root part is displayed. ALIASes are displayed in italic font to make them obvious. However, none of the MC78L05 symbols have a TO252 package so again I don't know what you mean.

Enzo1960, 2019-09-27:
You're right, LP2950 in TO252 package has 1 for input and 3 for output. But in symbol library 1 is marked "VO" and 3 is marked "VI".

Enzo1960, 2019-09-27:
EXCUSE ME!
I apologies, my failure: I choosed initially TO92 model and later I changed in my project footprint from TO92 to TO252, thinking it was same, No bug, excuse again waste of time!

closed 2019-09-27 by Enzo1960