KiCad/kicad-symbols #2249: USB ESD protection device TPD4EUSB30 "NC" pins are not truly NC

poeschlr, 2019-10-17:
The datasheet http://www.ti.com/lit/ds/symlink/tpd2eusb30a.pdf suggests connecting the NC pins in its suggested layout section meaning the pins should definitely not be invisible and arguably not even pin type "not connected". I would suggest to have the current NC pins placed as visible with pin type passive next to the signal pin that is suggested to be routed through it in the suggested layout section.

The datasheet for a similar device makes it even clearer http://www.ti.com/lit/ds/symlink/tpd4e05u06.pdf#page=16

NewFork17, 2020-05-31:
Hi Poeschlr,

I have also read your forum input. I think this is not a bug, rather a situation where TI does not apply enough information how to use the part related to the NC connection. Therefor even if possible KiCad can't provide a perfect solution for this.  The suggestion made in the forum about changing  the part according your own needs is the best answer I think.

I would consider to close this issue to clear up open items.

Kind regards,

evanshultz, 2020-08-24:
Also applies to `TPD4E02B04DQA` (https://www.ti.com/lit/ds/symlink/tpd4e02b04.pdf).