KiCad/kicad-symbols #2274: How do you install these libraries

thawkins, 2019-10-29:
Is there any step by step documentation on how to install these libraries in kicad, for linux uses.

DanSGiesbrecht, 2019-10-29:
http://www.kicad-pcb.org/libraries/download/ should provide the information you need. Does this answer your question?

DanSGiesbrecht, 2019-10-29:
Here's some more detailed information: https://forum.kicad.info/t/library-management-in-kicad-version-5/14636

poeschlr, 2019-10-29:
Most linux distros should come with a separate package for the libraries. I suggest to use that if you do not want to do something special.
https://forum.kicad.info/t/i-installed-kicad-5-under-linux-but-there-are-no-libraries-the-default-option-for-sym-lib-table-setup-is-disabled/12962

DanSGiesbrecht, 2019-10-30:
@poeschlr @thawkins, are you comfortable closing this issue?

poeschlr, 2019-10-30:
Yes as it is clearly a support question. The github library issues are not meant to be used for tech support

closed 2019-10-30 by poeschlr