KiCad/kicad-symbols #2293: Duplicated pins policy on arduino modules

5N44P, 2019-11-09:
Recently i found myself doing some new arduino symbols for a project, and i noticed that all boards have duplicated pins that are not stacked. 
Since the duplicated pins are shorted in the module, it doesn't make much sense to me to have them available for connecting different nets.

What do you think about this? Is it intentional? I would propose a PR to stack all stackable pins on the various boards, if you agree.

See for example the RESET and GND pins on the Arduino nano:
[image]

Or the GND and the double SDA/SCL in the Arduino uno:
[image]

poeschlr, 2019-11-09:
I do not think the reset pins should be stacked as the user might not always want to use both of them.

The positive supply pins are not stacked because one might want to show which decoupling capacitor goes with which pin. (might even want to use net ties to force a particular layout.)

The ground pins might however be best stacked.

5N44P, 2019-11-09:
The positive power supply pins are actually different: vin is a raw power input, 5v is a regulated 5V output (or input if vin is not connected), 3.3v is a power output, so they definitely should not be stacked. 

The reset should be pretty clear to the user that is the same signal, but the duplicate A4 and A5 on the Uno might be confusing because they have a double function, shouldn't it be clearer to the user that they are actually shorted in the module? Or is the users' responsability to understand that they are electrically connected in the module?