KiCad/kicad-symbols #2300: High pin count connector

daviid5, 2019-11-12:
Hello,

I'm currently designing a PCB with this connector:
https://b2b.harting.com/files/download/PRD/PDF_TS/0923148X921_100072926DRW058B.pdf
The connector has 48 pins.

Whats the best practice for an symbol for an connector with 48 Pins

[image]
Such an conector can't be the solution, it simply gets to large.

Do you have other suggestions?

David

evanshultz, 2019-11-12:
Our DIN 41612 footprints are nearly finished and then we will add symbols. Three-row symbols suited for this standard don't exist in the official library at this time. There is an issue about the symbol style to use but I'm not taking the time to search for it now.

You are free to make your own symbols. I assume you'll be making your own footprints.

For usability questions, https://forum.kicad.info/ is often best place to ask.

daviid5, 2019-11-12:
I've seen this style in the AppNote from TI:
[image]

Maybe this could be an solution?

Is there an PullRequest for the DIN 41612 connectors?

poeschlr, 2019-11-12:
For connectors with more than 3 rows a multi unit approach (one unit per row) might be the best option.

If we go with a single unit symbol then i suggest that a given row always exits on the same side of the symbol (I find it strange that in the screenshot half of row b exits to the left while the other half exits to the right)

cpresser, 2020-04-24:
Sometime there is also a default-pinout. In this case a specialized symbol does make sense.
Example: https://github.com/KiCad/kicad-symbols/pull/2613