KiCad/kicad-symbols #2345: GitLab Symbol Migration

sethhillbrand, 2019-11-27:
Hi Librarians-

As part of the Great KiCad GitLab migration of 2019, I am testing the movement of open issues and PRs from GitHub to GitLab.

Here's what we know so far:

- Open Pull Requests will transfer to GitLab without moving the source of the PR.  That means pull requests will become GitLab Merge Requests that are referenced to the user's GitHub repository
- Open Issues will transfer to GitLab issues
- Closed/Resolved PRs and Issues will also transfer

In order to facilitate the transfer, it will be helpful if you can do two things:
- Open a GitLab account (these can use your GitHub OAuth)
- Set your GitHub email to public

The last point is critical because the migration script will only link users whose e-mail addresses match.  If the script can't see your GitHub e-mail address, the issue will be tied to the KiCad-Bot that performs the migration.

The timeline for the actual migration will be determined by the librarian group but it would greatly assist my testing for bugs if you could do the two steps above as soon as possible.

Thank you-
Seth

calebreister, 2019-11-27:
> Open Pull Requests will transfer to GitLab without moving the source of the PR. That means pull requests will become GitLab Merge Requests that are referenced to the user's GitHub repository

Since I have a GitLab account already, is there a way I can migrate my KiCad library forks now? Will the bot be able to detect the GitLab fork and link to it instead of the GitHub version?

sethhillbrand, 2019-11-27:
We cannot migrate your branches. But as long as your GitHub email is public and the same as your GitLab email, the migrated PR will still be assigned to you and will reference your original GitHub branch.

aewallin, 2019-12-22:
Is there a status update on this? (last comment 25 days ago, not many days of 2019 left ;) )

are the new repos going to eventually appear here? https://gitlab.com/kicad/libraries

serkanxselcuk, 2020-02-18:
Hi librarians and developer team, @poeschlr @sethhillbrand 

Actually I find it great with structured subgroups, so every team can work under one kicad account as sub teams.
https://gitlab.com/kicad/libraries seems to be empty. Do you have any update for us? When will we move or will we move? is anyone working on it?

poeschlr, 2020-02-18:
I currently have no time to do it. I doubt it will change soon. The transition is a lot of work that does not really directly pay off as github and gitlab have basically the same feature set. (But different enough to make problems especially regarding the automated systems we have in place) It will also mean that most of the currently open pull requests will kind of die.

If you want to help then look into how the travis and CLA stuff could be transferred over to whatever the equivalent of it is on gitlab. Plus maybe help us with reviewing to get the number of open pull requests down (I am not going to lower our standards to achieve this so it can only happen with more reviewers)

chschlue, 2020-08-31:
Obsolete, closing.

closed 2020-08-31 by chschlue