KiCad/kicad-symbols #2367: Don't tag inverted input pins with ~ and N prefix

opatut, 2019-12-06:
Lots of components have pins that have `N` in the name to indicate a negative/not input (inverted), but are also tagged with `~` and thus receive a line above them to do the same. The result looks like this:

[image]

That implies an *inverted* `nRESET` pin, which would be an uninverted `RESET` pin. Basically this is a double negation, but it shouldn't be. There is multiple instances of `~n` in pin names, but not all of them are necessarily wrong, i.e. some of them might be correct because the `N` does not stand for negation but is part of the pin name. Here's a quick `grep` that shows all of them, I'm not sure which are wrong (BNO055 definitely is).

```
$ grep -rnFi 'X ~n'
obsolete/CPU.lib:52:X ~NMI 6 -500 -300 100 R 50 50 1 1 I
MCU_Module.lib:501:X ~NRST 14 -1300 3200 200 R 50 50 1 1 I
Sensor_Motion.lib:119:X ~nRESET 11 -600 500 100 R 50 50 1 1 I
Sensor_Motion.lib:139:X ~nBOOT_LOAD_PIN 4 -600 100 100 R 50 50 1 1 I
CPU.lib:1363:X ~NMI~ 17 -700 600 150 R 50 50 1 1 I
74xx_IEEE.lib:1028:X ~N 2 -600 250 300 R 50 50 1 1 I I
RF_AM_FM.lib:368:X ~nIRQ 5 -400 -200 100 R 50 50 1 1 O
Interface_USB.lib:155:X ~NOS 20 400 -600 100 L 50 50 1 1 I
Interface_USB.lib:348:X ~nXRES 14 -700 500 100 R 50 50 1 1 I
Interface_USB.lib:486:X ~nXRES 18 -700 700 100 R 50 50 1 1 I
Interface_USB.lib:1848:X ~NP3 24 -1000 0 200 R 50 50 1 1 I
RF_Module.lib:22:X ~NSS 15 -500 100 200 R 50 50 1 1 I
RF_Module.lib:641:X ~NRST 2 -1200 600 100 R 50 50 1 1 I
CPU_NXP_6800.lib:22:X ~NMI 6 -500 0 100 R 50 50 1 1 I
CPU_NXP_6800.lib:74:X ~NMI 2 -500 200 100 R 50 50 1 1 I
CPU_NXP_6800.lib:130:X ~NMI 2 -500 200 100 R 50 50 1 1 I
```

chschlue, 2019-12-06:
> MCU_Module.lib:501:X ~NRST 14 -1300 3200 200 R 50 50 1 1 I

NUCLEO144-F429ZI -> bug

> Sensor_Motion.lib:119:X ~nRESET 11 -600 500 100 R 50 50 1 1 I
> Sensor_Motion.lib:139:X ~nBOOT_LOAD_PIN 4 -600 100 100 R 50 50 1 1 I

BNO055 -> bug

> CPU.lib:1363:X ~NMI~ 17 -700 600 150 R 50 50 1 1 I

Z80CPU -> false positive

> 74xx_IEEE.lib:1028:X ~N 2 -600 250 300 R 50 50 1 1 I I

74173 -> disputable

> RF_AM_FM.lib:368:X ~nIRQ 5 -400 -200 100 R 50 50 1 1 O

ZETA-433-SO -> bug

> Interface_USB.lib:155:X ~NOS 20 400 -600 100 L 50 50 1 1 I

CH340T -> disputable

> Interface_USB.lib:348:X ~nXRES 14 -700 500 100 R 50 50 1 1 I

CY7C65211-24LTXI -> bug

> Interface_USB.lib:486:X ~nXRES 18 -700 700 100 R 50 50 1 1 I

CY7C65215-32LTXI -> bug

> Interface_USB.lib:1848:X ~NP3 24 -1000 0 200 R 50 50 1 1 I

TUSB2036 -> false positive

> RF_Module.lib:22:X ~NSS 15 -500 100 200 R 50 50 1 1 I

Ai-Thinker-Ra-01 -> bug

> RF_Module.lib:641:X ~NRST 2 -1200 600 100 R 50 50 1 1 I

MM002 -> bug

> CPU_NXP_6800.lib:22:X ~NMI 6 -500 0 100 R 50 50 1 1 I
> CPU_NXP_6800.lib:74:X ~NMI 2 -500 200 100 R 50 50 1 1 I
> CPU_NXP_6800.lib:130:X ~NMI 2 -500 200 100 R 50 50 1 1 I

false positives

Are you willing to patch these?

opatut, 2019-12-07:
> Are you willing to patch these?

I would be honoured. I am preparing a PR right now.

> 74173 -> disputable

The [datasheet](https://www.ti.com/lit/ds/symlink/sn54173.pdf) calls the inputs `N` and `M`.

[image]

They are technically being inverted inside the logic:

[image]

But IMO they should still be called `N` and `M`, without `~`, on the symbol. The fact that the interal logic uses the inverted signals of `N` and `M` doesn't mean that the signals are `nN` and `nM`. The only inputs with canonical inverted names are `~G1` and `~G2`. I'd patch this too.

> CH340T -> disputable

[Datasheet](https://cdn.sparkfun.com/datasheets/Dev/Arduino/Other/CH340DS1.PDF) calls it `NOS#` "active low" (the `#` seems to be a notation for active-low inputs). I'd keep the `~`.

[image]

closed 2019-12-29 by chschlue