KiCad/kicad-symbols #2446: Add open collector symbol

hildogjr, 2020-01-13:
Some parts / symbols listed above could have added the "open collector symbol". This will improve the schematics (main in a printed version):
7405 and 06 http://www.ti.com/lit/ds/symlink/sn74ls06.pdf
74596
7433
7403
...

evanshultz, 2020-01-14:
This topic is covered at https://github.com/KiCad/kicad-symbols/issues/2115, but apparently you _really_ want your own issue about it.

hildogjr, 2020-01-14:
Since this could be easy to fix than #2115 that is a more deep discussion, I use this to registry a pull request adding the "open collector" symbol to the logical parts.

hildogjr, 2020-04-08:
A implementation to better behavior is being discussed at https://gitlab.com/kicad/code/kicad/-/issues/4049