KiCad/kicad-symbols #2469: onfi nand flash

nerdyscout, 2020-01-28:
I am currently working in this symbol
[image]
based on this documentation
http://www.onfi.org/-/media/client/onfi/specs/onfi_4_0-gold.pdf?la=en

there are a few things i would like to ask before preparing a pull request:
- the symbol size is rather big as all pins are duplicate for both channels. do you prefer it the way on the picture or as independent units which would cut it down a bit?
- I would be fine to stack all GND pins, but regarding VCC and VCCQ I would like to keep them seperate to indicate the amount of decoupling capacitors. what do you think?
- the linked document is generic for various kind of flash types, these symbols could be applied to many different flashes from various manufacturers. Therefore it does need some generic naming scheme - any suggestions?

nerdyscout, 2020-02-04:
started now this pr #2477

cpresser, 2020-03-19:
Sorry for the late answer, perhaps it still helps you (or other users with similar questions):

> 
>     * the symbol size is rather big as all pins are duplicate for both channels. do you prefer it the way on the picture or as independent units which would cut it down a bit?

You can split it into units.

> 
>     * I would be fine to stack all GND pins, but regarding VCC and VCCQ I would like to keep them seperate to indicate the amount of decoupling capacitors. what do you think?

There are explicit rules for that. 
https://kicad-pcb.org/libraries/klc/S4.3/
TLDR: Don't stack the positive supply pins in this case, GND must be stacked.
> 
>     * the linked document is generic for various kind of flash types, these symbols could be applied to many different flashes from various manufacturers. Therefore it does need some generic naming scheme - any suggestions?

Make this one a fully specified symbol.
Other parts can be added as alias.

closed 2020-04-03 by nerdyscout