KiCad/kicad-symbols #2474: Footprint filter

jneiva08, 2020-01-29:
In some IC footprint filter have DIP and PDIP or SO and SOIC,
but i the datasheet have only SO or DIP, witch should choose or should be the two options?

chschlue, 2020-01-29:
Not sure if I understand the question correctly, but:
KiCAD FPs for plastic (P) and ceramic (C) DIP are both DIP while SO usually means 5.3mm wide SOP or 3.9mm wide SOIC, depending on manufacturer, moon phase and probably some other factors.
Hope this helps.

evanshultz, 2020-01-29:
Ultimately, what we are moving toward is having a unique symbol for each package. That will resolve any questions on this topic.

But in the meantime, and for some generic symbols, we have multiple footprint filters for physically-different packages. In this case, if the footprint filter is for a package that isn't available in a datasheet, I can imagine a few possibilities:
- A simple human error
- Some other vendor of the part offers the package but not the datasheet you saw
- The available packages were increased or reduced over time

Let us know if that isn't answering your question.

cpresser, 2020-04-24:
I am closing this since I consider the question answered. If not, please reopen.

closed 2020-04-24 by cpresser