KiCad/kicad-symbols #2504: More typs of switches

michalelektryk, 2020-02-21:
I'm beginner in KiCad so I may simply not be able to find them, but I have 4 and 6 position slide switches and they doesn't seem to be available in the library.  SS-14D01 for eg. All in all the switch library seems to be very basic not even including very popular 3DPT switch used in guitar effects.

Am I missing some libraries (maybe community based) or there simply switches aren't well supported?

evanshultz, 2020-02-21:
This library is largely driven by contributors. If you find useful symbols missing, please create a PR to add them. It would be great to have more symbols that KiCad users find beneficial. We do not take requests for symbols, though, because there is too much to do and we're all volunteers.

Please do search around in other PRs and issues here, take a look at http://kicad-pcb.org/libraries/klc/, and check the library, to get a feel for the symbol style and naming convention before you submit a PR. And ask if you have questions.

poeschlr, 2020-02-22:
We have a general rule about issues. We do not take any requests for assets. This is done as we can not reasonably expect that the library will ever have assets for all possible parts that exist. Issues are therefore reserved to report problems with existing assets.

As evan stated contributions are welcome but i will still close this issue. (For the future questions for library assets might be better asked on the forum.)

closed 2020-02-22 by poeschlr