KiCad/kicad-symbols #2514: 74HC137 and 74HC237 has the wrong pin name (A3 instead of A2)

norpan, 2020-02-28:
The symbol for these look like this:
[image]

chschlue, 2020-02-28:
Bug confirmed.
Also, 137's outputs aren't negated.

Are you willing to submit a PR to fix this?

norpan, 2020-02-29:
I can try, I have the modified one in my own library, but how would I in the easiest way put it in the github library?

norpan, 2020-02-29:
Shouldn't they just share the symbols with the other 74x137 and 74x138 circuits, I mean, they have the same pin layout as for instance 74LS137. Or is there something with the pin naming?

norpan, 2020-02-29:
Actually, looking through the 74xx library, it could do with a massive cleanup, it has different output names using letters Y, O, S etc. I guess different manufacturers will have different pin names, but since these are generic symbols, they should probably have the same naming standard.

Since TI was the original manufacturer of the 74 series, it makes sense to follow that naming and also to have the datasheet link to the TI datasheet. Unless there is some reason not to.

norpan, 2020-02-29:
I also see a lot of symbols using the circles (inverted input/output) even though the style guide says to use ~ for negated pins instead of the circle. Personally I like the circle, but I like consistency more :)

chschlue, 2020-03-01:
74LS137, 74HC137, 74LVC137 and so on can be aliases of each other (and already are currently, if I'm not mistaken). 137, 138, 237, 238 however cannot, 137 and 237 have latch pins, 237 and 238 aren't inverting.

The logic libs are in a terribly state, your comment isn't even scratching the surface.
The problem is, a real cleanup can only start after 5.1.6 is released because we cannot break existing designs during point releases. (Help is greatly appreciated)

Fixes like your original issue are possible and welcome.

norpan, 2020-03-01:
Ok, yes, I agree that backwards compatibility is needed. So basically the only thing that can be fixed are pin names? It would be great if there were some guidelines about which pin names to use for these generic circuits.

No, 74HC137 is not an alias of 74LS137. Perhaps the best fix would be to make it so. However, the 74LS137 has a different pin length, and also circles on the outputs, so doing that would not be backwards compatible.

For instance, for 74HC137/237, there are two manufacturers, and the pin names for the latch and output enable signals (pins 4,5,6) are like this:
Texas Instruments: LE, ~OE1, OE0
Nexperia: ~LE, ~E1, ~E2
Toshiba: ~GL, ~G2, G1

Anyway, I'll make a PR just changing the obviously wrong pin name, and the rest will have to wait. Let me know if you need help fixing some schematics later. (I guess when making a completely new version, all symbols could be regenerated using consistent rules and some kind of table input).

chschlue, 2020-03-01:
> Ok, yes, I agree that backwards compatibility is needed. So basically the only thing that can be fixed are pin names? It would be great if there were some guidelines about which pin names to use for these generic circuits.

While we have no written rule, I would grant the naming rights to TI, for the reasons you already stated above.

> No, 74HC137 is not an alias of 74LS137. Perhaps the best fix would be to make it so. However, the 74LS137 has a different pin length, and also circles on the outputs, so doing that would not be backwards compatible.

The most important thing is for pins to stay at the same position.
Adjusting pin lengths (and body outline accordingly) would be permissible. The questions is whether it's worth the effort at this time.

> Anyway, I'll make a PR just changing the obviously wrong pin name, and the rest will have to wait. Let me know if you need help fixing some schematics later. (I guess when making a completely new version, all symbols could be regenerated using consistent rules and some kind of table input).

In case you want to start fiddling around with the logic libs in the new (v6) file format, you can compile eeschema from https://gitlab.com/stambaughw/kicad/-/tree/sexpr-symbol-lib-plugin (https://gitlab.com/kicad/code/kicad/-/merge_requests/118).

norpan, 2020-03-01:
But changing pin length could lead to issues, right? I mean, a signal connected to a longer pin would no longer be connected if the pin was shortened.

I'm interested to learn about this new format, is it documented somewhere? Also, having all symbols in one single file in git is a bit fiddly when making patches. Is there a plan to make them separate files and then build them together? Or keeping them separate in KiCad? Perhaps using a zip-file of them? That would really make it easier to contribute as any new symbol would be a new file and any change to a symbol would not touch the gigantic library file.

chschlue, 2020-03-01:
> But changing pin length could lead to issues, right? I mean, a signal connected to a longer pin would no longer be connected if the pin was shortened.

Yes, that's why the outline rectangle has to be adjusted, too.
When you shorten a pin, its outside end stays where it was but the other end doesn't touch the body anymore.
 
> I'm interested to learn about this new format, is it documented somewhere? Also, having all symbols in one single file in git is a bit fiddly when making patches. Is there a plan to make them separate files and then build them together?

There have been lots of ideas, like expanding the alias system to some kind of inheritance model (so almost-but-not-quite-the-same symbols can inherit most properties from a common parent and change only what's required (like doc link, default FP, negated pins, ...)).
Unfortunately, @stambaughw hates persistent links to the format specs for some reason ;)

closed 2020-03-01 by chschlue