KiCad/kicad-symbols #2520: USB3 B connector has incorrect pin power type

dieterss, 2020-03-02:
The USB3 B type connector symbol annotates the VBUS & GND pins a "Power Input" whereas they should be "Power Output". The USB2 B type connector has this correct.
(Annotating these pins as "Power Output" has been described as the correct annotation in issue #989)

Much less interesting, but still seemingly incorrect, the connector drawing on the symbol reflects a type A connector instead of a B type connector. The USB2 B type symbol has this correct.

Lastly, a question: is there a reason the SS Tx/Rx lines are labeled Bidirectional instead of Input/Output ?

cpresser, 2020-03-19:
I was searching when this was introduced, but can't find it. Most likely this was copied over from the v4 libs in this commit a2bdaa3be670077b6daeeef81ddca099ffdd1a9d

@dieterss would you like to submit a fix for this issue? It seems like it is rather simple to fix and you already know the solution.

Regarding the Tx/Rx lines: No clue. Might be a Bug as well. The symbol `USB_C_Plug` also has them as BiDir.

diegoherranz, 2020-04-08:
On type-C, the TX/RX pairs are bidirectional because they can be repurposed for Alternate Modes using Power Delivery, for instance Display Port, and in that case the direction can change. I think those need to stay as bidirectional.

On the USB3_A and USB3_B, however, I think they could be defined as input/output.

PR #2655 (2020-09-17): Fix generic USB connector pin types
* Fix USB3_B drawing

Fixes #2520

[image]
[image]
[image]
[image]
[image]
[image]

---

All contributions to the kicad library must follow the [KiCad library convention](http://kicad-pcb.org/libraries/klc/)

Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [ ] Provide a URL to a datasheet for the symbol(s) you are contributing
- [ ] An example screenshot image is very helpful
- [ ] Ensure that the associated footprints match the [official footprint library](http [... cut]
commit 1726cb66 (2020-09-17): Fix generic USB connector pin types (#2655)

* Fix generic USB connector pin types (#2655)

* Fix USB3_B drawing

Fixes #2520

* Change USB-C and OTG-B power pins to passive
commit 69f1b5c6 (2020-09-22): Fix generic USB connector pin types (#2655)

* Fix generic USB connector pin types (#2655)

* Fix USB3_B drawing

Fixes #2520

* Change USB-C and OTG-B power pins to passive
closed 2020-09-17 by evanshultz