KiCad/kicad-symbols #2552: Analog_Switch ADG728 wrong Pin to signal assignment

kagehisa, 2020-03-16:
In the official Datasheet the pinout is the following:
[image]

In the Library the pinout for Signals 1 to 4 are correct but 5 to 8 are wrong.
[image]
Signal S8 should be on pin 9, S7 on pin 10, S6 on pin 11 and S5 on pin 12.
The other assignments are correct.

cpresser, 2020-03-19:
Can confirm, that looks broken.
Would you provide a fix for that? It is a rather simple change and a good way to start contributing to the library.

kagehisa, 2020-03-19:
Alright I will give it a try.

cpresser, 2020-03-19:
There are some instructions over here: https://kicad-pcb.org/libraries/contribute/

closed 2020-03-19 by chschlue