KiCad/kicad-symbols #2568: Analog_Switch CD4053B : wrong pinout

HelixOnGit, 2020-03-23:
Pin 14 and 15 on "Analog_Switch">"CD4053B" are swapped (in/out "X" and "Y", respectively).

The corresponding models contained in "IEEE_4xxx" and "4xxx" are correct. 

Application: Eeschema
Version: (5.1.5)-3, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

evanshultz, 2020-03-24:
@HelixOnGit 
I'm not seeing this. Maybe you can share what you're seeing.

Here is the current symbol:
[image]

And the pinout from http://www.ti.com/lit/ds/symlink/cd4051b.pdf:
[image]
[image]

It looks like the 'A' and 'B' switches have all the right pins. Am I missing something?

HelixOnGit, 2020-03-24:
That is strange. Please note my submitted version of kicad. I had an earlier version installed before I moved to 5.1.5 -- maybe there was a hiccup with the replacement/reinstall procedure? I haven't built from the sources, but used the win64 installer. 

The version you show is indeed correct. However I still get this picture: 
[image]

chschlue, 2020-03-24:
Fixed in https://github.com/KiCad/kicad-symbols/pull/2378

Why do we duplicate this in Analog_Switch, BTW?
I get that it is one, but I'd say it's first and foremost a 4000 series IC.