KiCad/kicad-symbols #2595: Analog_Switch ADG728 pin A1 should be input

ChaosinaCan, 2020-03-31:
According to the [ADG728 datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/ADG728_729.pdf), A0 and A1 are address select pins but A1 is currently bidirectional: 
[image]

These should both be inputs, right?

evanshultz, 2020-04-01:
@ChaosinaCan 
Yes, they should both be Input type. Thanks for taking the time to report this issue, but in this case I think you have an outdated library. https://github.com/KiCad/kicad-symbols/pull/2565 fixed this problem and I verified the current library is correct. Since there is no issue I'm going to close this issue, but thanks for letting us know and we hope to see you again!

closed 2020-04-01 by evanshultz

ChaosinaCan, 2020-04-06:
I was using the libraries from the latest KiCad installer, which looks to be pretty out of date. After checking out the latest code from GitHub I can confirm that this is fixed. Thanks!