KiCad/kicad-symbols #2612: Electrical symbols question

aris-kimi, 2020-04-06:
Hi everyone, 

I just saw [PR #514](https://github.com/KiCad/kicad-symbols/pull/514) about electrical motors, and i was thinking, is there a plan for a symbol library about electrical design? With items like circut breakers, switches, thermal protection and lots of others... Items without pcb footprint.

aris-kimi, 2020-06-12:
Well, i 've answered my own question.. I will upload a small libray i've started named something like IEC_Symbols, in a way i'm imagining, when it will be a little more complete, with generic symbols to be used for electrical designs.
I thought of this something like an extention for KiCAD usage.
Is this something i should proceed with? 
That is what i am asking now.. What do you think??
@evanshultz @myfreescalewebpage @cpresser

myfreescalewebpage, 2020-06-17:
@aris-kimi we have not a clear opinion, we spoke about that on https://github.com/KiCad/kicad-symbols/pull/1846. This PR is not really good because part are added separately, not in a single lib (the good way to do it). If you want to submit something, do it in a new lib.

aris-kimi, 2020-06-25:
Hi everyone, I want to learn your opinion about the lib structure in the PR presented.

@myfreescalewebpage ,
 i do have some thoughts about expanding KiCAD's usage field, but this will probably require some work from the dev team,
i present this image as a short description of the idea..
Question: Should i present my idea in the version 6 wishlist issue?
[image]

and this image is some sample from the lib created in the PR

[image]

myfreescalewebpage, 2020-07-05:
I have no opinion but will say : yes you can post to a wishlist issue to the Kicad development team..