KiCad/kicad-symbols #2623: Issues with QRE1113 and QRE1113GR symbols and footprint associations

saviocmc, 2020-04-09:
First of all, thank you all contributors very much for this awesome library!

Reference
-----------
The issues I found are related to the *Reflective Optical Sensors* **QRE1113** and **QRE1113GR** (same sensor with different leadform options), found in the library **Sensor_Proximity**.
Datasheet: https://www.onsemi.com/pub/Collateral/QRE1113-D.PDF

They are currently declared in the the [Sensor_Proximity.lib](https://github.com/KiCad/kicad-symbols/blob/master/Sensor_Proximity.lib) file [here (QRE1113)](https://github.com/KiCad/kicad-symbols/blob/166cde7a05b3e1077df4efe936f338f28066b231/Sensor_Proximity.lib#L395) and [here (QRE1113GR)](https://github.com/KiCad/kicad-symbols/blob/166cde7a05b3e1077df4efe936f338f28066b231/Sensor_Proximity.lib#L147)

And in the [Sensor_Proximity.dcm](https://github.com/KiCad/kicad-symbols/blob/master/Sensor_Proximity.dcm) file [here (QRE1113)](https://github.com/KiCad/kicad-symbols/blob/166cde7a05b3e1077df4efe936f338f28066b231/Sensor_Proximity.dcm#L87) and [here (QRE1113GR)](https://github.com/KiCad/kicad-symbols/blob/166cde7a05b3e1077df4efe936f338f28066b231/Sensor_Proximity.dcm#L93)

Issues
------
- The descriptions and footprint associations in the symbols are swapped. According to the [datasheet](https://www.onsemi.com/pub/Collateral/QRE1113-D.PDF), the QRE1113 is the Through Hole version and the QRE1113GR is the SMD counterpart. This description and footprint associations are currently inverted in the lib.

- The QRE1113GR is defined as an alias of the ITR8307-F43 [(*datasheet*)](https://everlighteurope.com/index.php?controller=attachment&id_attachment=5385), but the two devices have different pin number definitions. For instance, the *Anode* and *Cathode* pins of the IR LED are inverted in the two devices.

  - QRE1113
[image]

  - ITR8307-F43
[image] 

  - *Obs:* The pin *numbers* of the two devices is different but the pin *layout* of QRE1113 and ITR8307-F43 are identical.

- The same problem is present in the definition of an alias of the QRE1113, the ITR1201SR10AR [(*datasheet*)](http://www.everlight.com/file/ProductFile/ITR1201SR10AR-TR.pdf).

Questions
-----------
1 - Is it OK to have an alias of a device with **different** pin number definition but same pin layout?
I personally think this is a great source for confusion.

2 - The QRE1113 [datasheet](https://www.onsemi.com/pub/Collateral/QRE1113-D.PDF) defines a third device, the QRE1114GR (identical to the QRE1113GR). I was not able to find it in the main component marketplaces or anywhere else in the market. Should it be added as an alias of the QRE1113GR anyway?

Suggestions
-------------

- First of all, the **Value field** of the QRE1113 and QRE1113GR symbols **MUST** be swapped, as it is currently a direct contradiction of the [datasheet](https://www.onsemi.com/pub/Collateral/QRE1113-D.PDF).

- Create dedicated individual symbols for each of the components: QRE1113, QRE1113GR, ITR8307-F43 and ITR1201SR10AR to **not** define them in terms of alias of each other, as **NONE** of them have, at the same time, the same pin numbers/layout and same footprint.
  - *Obs:* The [BPR-105F](http://www.ystone.com.tw/en/data/goods/IRPT/Photo%20Interrupters-Reflective%20Type.pdf) can continue be an alias of the ITR8307-F43 as they have the same pin numbers and same footprint.

- Create a THT footprint for the QRE1113 based on the footprint of the ITR8307-F43 but with the appropriate pin numbers.

I may submit a PR in a few days implementing these suggestion.

chschlue, 2020-04-09:
> ## Questions
> 1 - Is it OK to have an alias of a device with **different** pin number definition but same pin layout?
> I personally think this is a great source for confusion.

I'm inclined to say that Everlight's numbering is definitely asinine (pin 2 mark anyone?).
That being said, we already have similar cases. The TO-92 footprint is numbered left to right, for example, while some manufacturers number their TO-92 parts right to left, leading to apparently wrong symbols.
In this case, I'd vote for adding an Everlight-specific FP using Everlight's numbering (because it's not a standard package anyway.

> 2 - The QRE1113 [datasheet](https://www.onsemi.com/pub/Collateral/QRE1113-D.PDF) defines a third device, the QRE1114GR (identical to the QRE1113GR). I was not able to find it in the main component marketplaces or anywhere else in the market. Should it be added as an alias of the QRE1113GR anyway?

If ON claims to offer that part, it's probably permissible to add it. But what's the point?

saviocmc, 2020-04-09:
@chschlue thanks for the quick answer.

> In this case, I'd vote for adding an Everlight-specific FP using Everlight's numbering (because it's not a standard package anyway.

I agree. The [current SMD FP](https://github.com/KiCad/kicad-footprints/blob/master/OptoDevice.pretty/OnSemi_CASE100CY.kicad_mod) assigned to the SMD version of these devices is using the ON numbering (with pin 1 mark), so another one should be created for the ITR1201SR10AR using the Everlight's numbering.

On the other hand, the [current Through Hole FP](https://github.com/KiCad/kicad-footprints/blob/master/OptoDevice.pretty/Everlight_ITR8307F43.kicad_mod) assigned is using the Everlight's numbering. So another one should be created using the ON numbering for the QRE1113.

Do you agree with the creation of these two additional footprints?

Also, do you agree with this suggestion?

> - Create dedicated individual symbols for each of the components: QRE1113, QRE1113GR, ITR8307-F43 and ITR1201SR10AR to not define them in terms of alias of each other, as NONE of them have, at the same time, the same pin numbers/layout and same footprint.

chschlue, 2020-04-09:
LGTM

chschlue, 2020-05-08:
@saviocmc Are you still on this?

saviocmc, 2020-05-08:
@chschlue I got busy the past few weeks, but can I manage to submit a PR this weekend.

chschlue, 2020-09-20:
@saviocmc Will you be able to wrap this up by Tuesday?

saviocmc, 2020-09-20:
Sorry about the delay. I'm going to get this done by tomorrow.

PR #3086 (2020-09-23): Fix inconsistences between On Semi QRE1113/QRE1113GR and Everlight ITR8307/ITR1201SR10
Fix #2623 : some inconsistencies in the Sensor_Proximity library.
----------------------------------------------------------------------

Some of the added symbols depends on the merge of this pull request:
https://github.com/KiCad/kicad-footprints/pull/2482

## Issue summary

Both OnSemi and Everlight provide very similar (identical) proximity reflective sensors.
Both provide the sensors as SMD or THT devices.
The main difference is in the way each chose to assign the pin numbers.
I notice that the library  [... cut]
closed 2020-09-23 by chschlue