KiCad/kicad-symbols #2656: 5.1.6 release date will be May 15th

poeschlr, 2020-04-19:
See source code freeze announcement https://lists.launchpad.net/kicad-developers/msg43665.html

---

Please add pull requests as comments that you want to be included such that we know where to focus our resources.

I personally try to get around to finally check out https://github.com/KiCad/kicad-footprints/pull/1586

chschlue, 2020-04-19:
I usually add bugfixes and such to the next milestone to keep track.
Isn't that what we have them for?

Another question: kicad-source on GL has a 5.1.7 milestone but I couldn't find anything about another planned 5.1 release on the dev mailing list. Do you know more?

stambaughw, 2020-04-19:
There will most likely be a 5.1.7 release before V6 is release.  We are hoping to implement the V6 feature freeze by KiCon 2020 and release by the end of the year.  This will leave plenty of time to release another 5.1 branch bug fix between now and then.

chschlue, 2020-04-19:
This also leaves little time to migrate the libs to GL, to v6 and get rid of quite a bunch of technical debt.

evanshultz, 2020-04-19:
Yeah, I would like to target resolving all known bug issues and PRs also for each release. Status now:
- Symbol issues [15]: https://github.com/KiCad/kicad-symbols/issues?q=is%3Aopen+is%3Aissue+label%3ABug
- Symbol PRs [8]: https://github.com/KiCad/kicad-symbols/pulls?q=is%3Aopen+is%3Apr+label%3ABug
- Footprint issues [20]: https://github.com/KiCad/kicad-footprints/issues?q=is%3Aopen+is%3Aissue+label%3ABug
- Footprint PRs [8]: https://github.com/KiCad/kicad-footprints/pulls?q=is%3Aopen+is%3Apr+label%3ABug
- 3D issues [2]: https://github.com/KiCad/kicad-packages3D/issues?q=is%3Aopen+is%3Aissue+label%3ABug
- 3D PRs [1]: https://github.com/KiCad/kicad-packages3D/pulls?q=is%3Aopen+is%3Apr+label%3ABug

Note that some of the symbol items are meant for the next major release. Some didn't have a milestone assigned and I didn't take the time to look at each one.

@poeschlr is there more work to be done at https://github.com/KiCad/kicad-footprints/issues/1410? It's unclear if this has been resolved as there is only one linked PR and it was merged.

https://github.com/KiCad/kicad-footprints/pull/2166 is a bug, I guess, though it's not marked as one. But it's minor and should be easy.

https://github.com/KiCad/kicad-footprints/pull/1076 is a very nice addition that I would like to see added. There have been users asking about DIN 41612 footprints and other handmade PRs for them so it seems to me the community would like these footprints merged as well. Once that's done we'll have to look at symbols...

And I selfishly would like to see https://github.com/KiCad/kicad-footprints/pull/1145 get merged. @diegoherranz has posted a few times lately so perhaps we will get an update soon if this PR is ready or if someone else should take it over?

diegoherranz, 2020-04-19:
> And I selfishly would like to see KiCad/kicad-footprints#1145 get merged. @diegoherranz has posted a few times lately so perhaps we will get an update soon if this PR is ready or if someone else should take it over?

Let me have a look at https://github.com/KiCad/kicad-footprints/pull/1145. I should have some time now.

evanshultz, 2020-04-20:
@poeschlr 
I also consider https://github.com/pointhi/kicad-footprint-generator/pull/438 to be a bug. We are claiming IPC7351 footprints, using IPC-SM-782 size definitions where available, but I found that R, L, and C parts have different body sizes which the generator was assuming they were all the same. If you could check the script and confirm the changes are OK, I will resolve conflicts on that script PR and submit the updated footprints. I really wanted your eyes on the script first.

poeschlr, 2020-04-24:
@evanshultz i think the EP size stuff has not yet been started at all. A few footprints will therefore still have the wrong sizes. (Wrong in the sense of they use the maximum size of the datasheet but the size agrees with the footprint name so the user sees it easily)

chschlue, 2020-04-24:
I think https://github.com/KiCad/kicad-footprints/pull/2072 would be ready to go if it weren't for https://github.com/pointhi/kicad-footprint-generator/pull/474.
@poeschlr since you merged that one, what are we going to do with it? (see also https://github.com/KiCad/kicad-footprints/pull/2072#issuecomment-579425699)

poeschlr, 2020-04-25:
I introduced a minor fix for unifying the pin 1 markers of DFN footprints https://github.com/KiCad/kicad-footprints/pull/2244

poeschlr, 2020-04-26:
@evanshultz your script changes for differentiating R/L/C sizes is generally ok. However, a final verdict requires checking if only the intended footprints change when running the script. (Which means a footprint pull request is required)

poeschlr, 2020-04-26:
Oh and if i can be selfish, i would like the solder wire stuff (https://github.com/KiCad/kicad-footprints/pull/2062) to be merged soon as it would allow me to simplify two tutorials on the forum.

evanshultz, 2020-04-26:
KiCad/kicad-footprints#2244 has been merged.

@poeschlr 
I left comments at pointhi/kicad-footprint-generator#438 and, once those are resolved, will make a footprint PR. Thanks for taking a look at that one!

chschlue, 2020-05-09:
State of the libs with a week to go:
https://github.com/KiCad/kicad-symbols/milestone/11 8 open, 1 unassigned, 1 assignee possibly inactive
https://github.com/KiCad/kicad-footprints/milestone/7 10 open, 3 unassigned, 1 assignee possibly inactive
https://github.com/KiCad/kicad-packages3D/milestone/6 0 open
https://github.com/KiCad/kicad-templates/milestone/5 1 open, 1 unassigned

(I added everything mentioned here so far to these milestones and selfishly added https://github.com/KiCad/kicad-templates/pull/29, which isn't terribly urgent but should be easy to review (if you happen to have a particular skillset @poeschlr @cpresser))

Templates seem to be our red-headed stepchildren...

poeschlr, 2020-05-11:
Yesterday @stambaughw posted to the mailing list that everything shall be tagged by today such that the packagers have a week to get everything done. I will now check the symbol libs for problems and if there are no major design breaking changes then i will create the tag.
The remaining issues will have to wait for 5.1.7 or for version 6 depending on how exactly we decide to handle the next few months.

chschlue, 2020-05-11:
@poeschlr please give me a couple minutes to check something that just came up today.

poeschlr, 2020-05-11:
Is ok, just ping me (i will still check what we have so far, your change will hopefully not fully change the lib)

chschlue, 2020-05-11:
Not a huge deal, really. Just https://github.com/KiCad/kicad-symbols/pull/2718.
After https://github.com/KiCad/kicad-symbols/pull/2539 got merged, https://github.com/KiCad/kicad-symbols/pull/2708 which adds the rest of the family brought about that one symbols was named wrong.
Would be a pity if we we were stuck with if it's released, that's all.

chschlue, 2020-05-11:
Sry, ping @poeschlr

chschlue, 2020-05-11:
> Also technically we have until midnight ;) (and who says we need to use CET here)

I would always assume UTC if no timezone is given. We probably even have until the early morning, @stambaughw's midnight is pretty late, I think ;)

poeschlr, 2020-05-11:
# Analysis of Symbol lib
## Removed (or renamed) symbols
- `Interface_USB.lib:FT2232H`
  - Split QFP and QFN version https://github.com/KiCad/kicad-symbols/pull/2420
- 555 clean up https://github.com/KiCad/kicad-symbols/pull/2429
  - `Timer.lib:LM555` 
    - (and its aliases `ICM7555`, `MC1455`, `TLC555CD`, `TLC555CP`, `TLC555CPS`, `NA555`, `NE555`, `SE555` and `SA555`)
  - `Timer.lib:TLC555`
  - `Timer.lib:TLC555CPW`

## Wrongly reported as removed
- `power.lib:~Earth`
- `power.lib:~Earth_Clean`
- `power.lib:~Earth_Protective`

For these the ~ was removed. I am not sure what that change exactly means. See PR https://github.com/KiCad/kicad-symbols/pull/2592

# Pins have been moved, renumbered or removed in symbol
- `74xx_IEEE.lib:74LS139` Bugfix (had wrong pin numbers) https://github.com/KiCad/kicad-symbols/pull/2654
- `Analog_Switch.lib:CD4053B` Bugfix (wrong pin numbers and names) https://github.com/KiCad/kicad-symbols/pull/2378
- `Converter_DCDC.lib:TC7662Bx0A` and `TC7662BxPA` Bugfix (pins inside symbol) https://github.com/KiCad/kicad-symbols/pull/2320
- `Driver_Motor.lib:DRV8662` Enhancement (stacked power pins and shortened pins) https://github.com/KiCad/kicad-symbols/pull/2407
- `Driver_Motor.lib:TB6612FNG` Bugfix (stack output pins) https://github.com/KiCad/kicad-symbols/pull/2677
- `MCU_SiliconLabs.lib:EFM8BB10F8G-A-QFN20` Enhancement (stacked GND) https://github.com/KiCad/kicad-symbols/pull/2545
  - and its aliases `EFM8BB10F2A-A-QFN20`, `EFM8BB10F2G-A-QFN20`, `EFM8BB10F2I-A-QFN20`, `EFM8BB10F4A-A-QFN20`, `EFM8BB10F4G-A-QFN20`, `EFM8BB10F4I-A-QFN20`, `EFM8BB10F8A-A-QFN20`, `EFM8BB10F8I-A-QFN20`
- `Power_Protection.lib:PESD3V3L4UG` and alias `PESD5V0L4UG` as well as `PESD3V3L4UW` and alias `PESD5V0L4UW` Bugfix (pinnumbers) https://github.com/KiCad/kicad-symbols/pull/2619
- `Power_Protection.lib:USBLC6-2SC6` Enhancement (pin lengths) https://github.com/KiCad/kicad-symbols/pull/2598/files
- `RF.lib:HMC431` Bugfix (NC pins are suggested to be connected) https://github.com/KiCad/kicad-symbols/pull/2661
- `RF.lib:SE5004L` Enhancement (stack GND pins as they are connected in the footprint) https://github.com/KiCad/kicad-symbols/pull/2659

# Normal pins ok but NC pins have been moved, renumbered or removed in symbol
- `Converter_ACDC.lib:IRM-03-3.3` and its aliases `IRM-03-5`, `IRM-03-12`, `IRM-03-15`, `IRM-03-24`

---

Full changelog
[5.1.6_vs_5.1.5.txt](https://github.com/KiCad/kicad-symbols/files/4611402/5.1.6_vs_5.1.5.txt) 
[5.1.6_vs_5.1.5.log](https://github.com/KiCad/kicad-symbols/files/4611403/5.1.6_vs_5.1.5.log)

evanshultz, 2020-05-11:
The 'Wrongly reported as removed' section is OK. See comments in the PR.

I feel it's a shame that we now have nice DIN 41612 footprints but not corresponding symbols. If the library can wait to be tagged for 24 hours I will plan to generate them by then.

Also, I would really like to get https://github.com/pointhi/kicad-footprint-generator/pull/438 merged since it's a bug, IMO, but I haven't had time to try and figure out how to untangle things.

poeschlr, 2020-05-11:
There will always be things that would be nice to get in especially shortly before a release. There will be future releases so the work will not be lost ;) 
I will try to help you with untangling the footprint generator fixes sometime this week. Depends a bit how stressful work is. (One of my responsibilities at work is to fixup git problems of others so i have gotten a lot of experience lately)

By the way the reason why i did not warn you guys yesterday was because i was seeing my family for the first time since mid February which had a bit higher priority than KiCad. And today i needed to work.

nickoe, 2020-05-11:
@poeschlr If you think some hints are needed for the release announcement about the library, please add them to https://github.com/KiCad/kicad-website/blob/master/content/blog/release-5.1.6.adoc

stambaughw, 2020-05-11:
Nick,

The blog post title still says 5.1.5.

Cheers,

Wayne

poeschlr, 2020-05-11:
@nickoe if you want you can link to the diff logs i also link in the release page (for symbols only). But i am not aware that we did that for past releases.

nickoe, 2020-05-11:
@poeschlr We have done it for some. If you (the librarians) want to and have something semiautomated it could make sense to show it at every release. For example in https://kicad-pcb.org/blog/2017/03/KiCad-4.0.6-Stable-Release/ we listed some renames... If you don't think it makes sense in this case, we will just stick with the generic point release format as we have now.

nickoe, 2020-05-12:
@poeschlr @cpresser Related to this is  https://lists.launchpad.net/kicad-developers/msg43785.html

closed 2020-06-21 by poeschlr