KiCad/kicad-symbols #2732: Temperature_Sensors -> MCP9804_DFN has wrong package

svollenweider, 2020-05-15:
The MCP9804_DFN has the wrong package (3x3 DFN). The correct package would be 

Package_DFN_QFN:DFN-8-1EP_3x2mm_P0.5mm_EP1.3x1.5mm

chschlue, 2020-05-15:
Confirmed.

The current package is completely wrong. However, neither your suggestion nor `DFN-8-1EP_3x2mm_P0.5mm_EP1.75x1.45mm` which we have for [Microchip's MC package](http://ww1.microchip.com/downloads/en/DeviceDoc/8L_DFN_2x3x0_9_MC_C04-123C.pdf) is completely correct.
Both are within the EP tolerance and so are both acceptable, but the best solution would be to generate a new footprint using the generator at https://github.com/pointhi/kicad-footprint-generator/tree/master/scripts/Packages/Package_NoLead__DFN_QFN_LGA_SON. Would you be willing to do that?

svollenweider, 2020-05-15:
Hey
I added a package with a 1.4mm x 1.7mm Base under this pull request

https://github.com/KiCad/kicad-footprints/pull/2284

chschlue, 2020-05-15:
Yes, absolutely.
In fact, we generally trust the generator's correctness and actually check its input instead of its output to review scripted FP additions.

svollenweider, 2020-05-15:
Here you go

chschlue, 2020-08-04:
FP merged. Are you going to update the symbol, too?

PR #3089 (2020-09-23): MCP9804_DFN fix package
Closes: #2732
closed 2020-09-23 by chschlue