KiCad/kicad-symbols #2761: ADC LTC1864/LTC1865 and LTC1864L/LTC1865L

dlcflv, 2020-05-29:
There is a (very small, indeed) error in the description of these devices in the Analog_ADC.dcm file.
These are SAR (successive approximation), not sigma-delta, ADC.

cp-aquila, 2020-08-04:
Good find!
That seems like an easy fix. Are you up to provide a patch and open a PR?
I think there is also a type in the description, it says `Digial`

dlcflv, 2020-08-07:
I am really a beginner in Kicad and do not know at all the procedures for opening a pull request. Although I know what changes to make to the two files Analog_ADC.lib and Analog_ADC.dcm, I don't know how to do it. Personally I got around by creating my own library (as well as when I have to create a new component)

aris-kimi, 2020-08-07:
Hi @dlcflv , you can take a look here: https://kicad-pcb.org/libraries/contribute/ for the basic procedure. And of course ask for something when you are not sure. I think you can get this through. It is simpler than it might seems. First steps are a bit tricky though..