KiCad/kicad-symbols #2768: STGIPSxxxxx Design help

aris-kimi, 2020-05-30:
Hi everyone.

I'm about to add thoses ic's but i need some guidance about an appropriate symbol appearance.. I think from those three topologies the first is the best, or the third...

[image]

evanshultz, 2020-05-31:
https://www.st.com/resource/en/datasheet/stgips10c60.pdf is one datasheet.

The first will make the bootstrap caps awkward to wire up.

The last is the best of the above, IMO. But putting CIN on the right side would make it even better. And the OUTx pins could be Output type.

Additionally, page 14 of the datasheet above is another pin layout that looks good to me. I might even like it better than option three above.

Submitting a snippet of a schematic using the symbol in your PR might be a good way to show that the pin layout works nicely.

aris-kimi, 2020-05-31:
I have made some changes to all so they can easily be exchanged. I think i should create a new lib though, named IGBT_Module, and put them in there instead of Driver_Motor, is this ok??

[image]

closed 2020-09-30 by aris-kimi