KiCad/kicad-symbols #2856: Electrical type of DWM1000 VSS pins should be Power input, not Power output

gavanfantom, 2020-07-11:
The schematic symbol for DWM1000 in RF_Module has all of the VSS pins set to the electrical type Power output. This causes ERC to fail, as these pins all need to be tied together.

The correct setting for all of the VSS pins is Power input.

cpresser, 2020-08-04:
Would you be so kind and create a patch for that and open a PR. Then we can merge the fix upstream.

gavanfantom, 2020-08-13:
This pull request sorts out the VSS pins and the I/O pins for the DWM1000. I had a go at giving the DWM1001 a similar treatment, but it seems to require more than a quick go over with a text editor, so I reverted most of my changes to that one. It should still be a (small) incremental improvement.

PR #2912 (2020-09-03): Correct pin types for DWM1000 and fix a few for DWM1001
* All ground pins on DWM1000 are now power inputs
* I/O signals corrected on DWM1000
* One or two signals corrected on the DWM1001, but the GND stack left wrong to appease the automated checker.

Fixes #2856

---

All contributions to the kicad library must follow the [KiCad library convention](http://kicad-pcb.org/libraries/klc/)

Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [ ] Provide a URL to a datasheet for the symbol(s) you are contr [... cut]
closed 2020-09-03 by myfreescalewebpage