KiCad/kicad-symbols #2892: ATSAMD21EXX-M EP should be connected to GND

marekr, 2020-08-01:
The SAMD21 G and J do not have their exposed pads connected to anything internally.

However, Atmel put a small foot note hidden in the packaging sheet explicitly for 32 pinners.

[image]

The E sized symbols currently have the EP pin set as "NC" so its hidden.

cp-aquila, 2020-08-04:
@0xdec Do you want to take a look at this?

0xdec, 2020-08-05:
@cp-aquila I think I remember seeing this when I made the symbols. My thinking is that we could go either way. Just because the pad is internally connected to GND does not mean that they actually want us to connect it to GND on the board. This might be a time for contacting a Microchip field applications engineer.

0xdec, 2020-08-05:
I'll also try to find some reference designs to see what they do

cp-aquila, 2020-08-06:
There are two different variants of the `32 lead QFN` footprint in the datasheet. One as ` with Sawn Wettable Flanks`.
Only one of them has states

> The exposed die attach pad is connected inside the device to GND and GNDANA

Section 3.0 and 3.1 of the Datasheet have more information on the 'wettable flanks' thing. All `SAMDA1xxx` seem to have them. For the `SAMD21xxx` I am not sure how to read it. Does all the info only apply to the AEC-Q100 parts?

One reference I found just adds more confusion. The Adafruit M0-Feather uses the QFN-48 Footprint. And has EP connected to GND:
[image]

cp-aquila, 2020-08-06:
See also
https://microchipsupport.force.com/s/article/SAMDA1-QFN32-package-exposed-pad

And the pretty generic answer (it depends):
https://microchipsupport.force.com/s/article/Where-to-connect-the-exposed-pad-of-a-QFN-package