KiCad/kicad-symbols #2895: fairechildsemi.com links for datasheet are leading to a page that is not a pdf file

AymenLA, 2020-08-04:
exemple : 
http://www.fairchildsemi.com/ds/LM/LM78L05A.pdf 
takes you to 
https://www.onsemi.com

this behavior is seen in linkes provided in "Regulator_Linear.dcm", I didn't check other symbol files.

chschlue, 2020-08-04:
Yes, ON Semi was too lazy to set up decent redirects after purchasing Fairchild.
Fixes are welcome.

AymenLA, 2020-08-04:
are links pointing to "alldatasheet.com" acceptable as a fix, or should we use only those provided by the manufacturer ?

chschlue, 2020-08-04:
Manufacturer-provided datasheets are preferred (if available).
Next best would be large distributors (like Mouser, for example).

Please use dubious websites only as a last resort. (Alldatasheet is _kind of_ trustworthy, I guess)

AymenLA, 2020-08-05:
Hi,

I might have a fix for the symbol files.
how do I submit my changes ? 
I've created a branch with this issue number but when I do a "git push" I get error 403 
"remote: Permission to kicad/kicad-symbols.git denied to AymenLA." ( this is normal .. )

how do I proceed to get the permission to push my branch ?
BR,

Aymen

chschlue, 2020-08-05:
https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

AymenLA, 2020-08-07:
Hi,

here is the pull request :  [https://github.com/KiCad/kicad-symbols/pull/2902]

BR,
Aymen

AymenLA, 2020-08-07:
Hi,

I've made a mistake regarding my last pull request : some changes where made and committed with my generic username and I didn't pay attention to that, this caused a problem for signing the CLI with that username, I've closed the pull request and I've made a new one.

issue-2895 correcting fairchildsemi.com dead links #2903 

BR,
Aymen

AymenLA, 2020-08-08:
Hi,
just committed the change request,
BR,
Aymen

closed 2020-08-08 by chschlue