KiCad/kicad-symbols #2919: Make "R", "C" etc optionally link to "_small" symbols

davidsrsb, 2020-08-19:
I know that I am not the only one who uses symbols R_Small, C_Small etc most of the time
When the symbol libraries get reorganized, why not make all the single letter default V5 symbols R, C, D etc R_Large and so on and then make the single letter shortcuts symbolic links to Small or Large as a user preference set in the library browser.
This would save a lot of key presses in drawing a schematic.
The actual symbol used would be R_Large or R_Small "R" being a logical shortcut set by a preference
This would make projects easy to share between users with either preference as the same library is used by both groups.

I posted this originally on GitLab issues as a wishlist as it would involve a change to the symbol browser, but it was closed as a library issue

chschlue, 2020-08-19:
I admit to being a _Small user, too, and I like the idea.

That being said, I have no idea how this would be a library issue, except for possibly renaming R, L, C to *_Large.

stambaughw, 2020-08-19:
I think we have to move them to a separate library because changing `R` and `C` symbols in the `Device` library will cause legacy (<=5.1.x) schematics to remap every `C` and `R` symbol which can be risky.  Far too many users delete the library cache file.

poeschlr, 2020-08-28:
@stambaughw I am not sure we can protect users who delete the cache lib if we want the lib to be able to evolve. We were ok with limiting our development during a minor release but am not so sure it would be a good idea if we even limit ourselves over a major release border. (However it is no longer my call to make so you will need to discuss this with @chschlue in the end)

How about adding good documentation how a user could use an installation of the old lib to get back a project they damaged by wrongfully deleting an important part of the kicad project instead of limiting our progress? (After all the v5 lib will stay available)

After all if we can not change the device lib the same can be said about every other lib as well. And that would mean we can not do anything that we already pushed back to v6 in order as to not bring user projects in danger with a minor release. Other planned examples include removing hidden power pins of the remaining logic libs and reorganizations in general.

poeschlr, 2020-08-28:
@davidsrsb While I am not against changes to the device lib I still am not sure if the current feature set would allow for fulfilling your request. I at least am not aware about a feature we could use to facilitate this linkage. The best i can think of is leaving the single letter versions unused in the official lib and allowing users to have them defined in their personal libs.

To be honest a better option (short term) might be if users libraries can be prioritized in the "Add symbol" dialog allowing users to basically redefine the use of these single letter variants in their own libs, having them match first while not requiring changes to the lib setup. (I seem to remember this option being mentioned on the forum some time ago)

chschlue, 2020-08-28:
> Edit: again @chschlue I hope you are ok with me voicing my opinion here. It will possibly take me some time to learn how to best fit into the role of a pure maintainer.

Of course!
In fact I'm pondering allowing even the rabble to voice their opinion.
Perhaps I'll be proclaiming the KiCad glasnost epoch some time this weekend ;)

But back on topic: I recall a GitLab issue concerning transforming the De Morgan alternatives into full-fledged alternative symbol styles. Searching for it, GL keeps throwing server errors ATM so I don't know whether this entails much more than just not calling it De Morgan anymore because it doesn't matter if the alternate style is in fact a logic equivalent anyway.

chschlue, 2020-08-28:
Apart from that, I agree with @poeschlr that mollycoddling users that can't keep their hands off the cache lib can't be our main concern (it's quite well documented I think.) At least not during a major release.

stambaughw, 2020-08-28:
@poeschlr I wasn't suggesting that we not make changes to the Device library but users tend to forget or are not aware of the ramifications of doing so.