KiCad/kicad-symbols #2926: Advice on symbol library management for contributing

mattkefford, 2020-08-24:
Hello all! I would like to add a component for a 1-input level shifter, it seems it would be easier to just add to my own library but nicer to add to the KiCad library. My github knowledge is a bit rusty though.

I have read the [guidelines](https://kicad-pcb.org/libraries/contribute/) and it mentions to fork and clone the repo. When KiCad is installed doesn't it already clone to repo to '/Library/Application Support/kicad/library'? I was under the impression that the library was updated weekly but I went to that folder, ran 'git status' and got "not a git repository".

So if I clone the repo do I clone it to /Library/Application Support/kicad/library and replace the one that's there? 

Thanks!

chschlue, 2020-08-24:
I am not very familiar with macOS, but I would generally advise against overwriting the libs installed system-wide.
Better clone the repos somewhere in your home dir and change environment variables (Preferences->Configure Paths) to point there.

mattkefford, 2020-08-24:
OK thank you, I thought that might be the safer approach but it is less clean and causes duplication. 

This page answers quite a few of my questions! https://forum.kicad.info/t/library-management-in-kicad-version-5/14636

closed 2020-08-24 by mattkefford