KiCad/kicad-symbols #2965: GITHUB LOCKDOWN AND TRANSFER TO GITLAB

cederom, 2020-09-01:
Hello world :-)

I can see there is a lockdown of the repo because of v6 preparations.. and the repository will be transferred to GITLAB.

Where should I put my current work? On GitLab??

The new GitLab repository does not seem to be there yet: https://gitlab.com/kicad/libraries/kicad-symbols/ returns 404..

Best regards :-)
Tomek

evanshultz, 2020-09-01:
Hello.

PRs now get a message around this, for example your PR at https://github.com/KiCad/kicad-symbols/pull/2964. I assume that's what you're referring to.

For posterity and anyone else who seems this, here is the text:
[image]

There is still work to be done on the librarian side for the GitLab migration, but all the instructions you need as a user are (hopefully) there. If you have set up your GitLab account then everything should be moved to GitLab automatically; you can then continue on GitLab or re-submit a PR.

cederom, 2020-09-01:
Allright then, waiting until 20201001 KiCAD repositories are moved from GitHub to GitLab. Good choice :-)

closed 2020-09-10 by myfreescalewebpage

myfreescalewebpage, 2020-10-01:
@rjmaris will be done by the end of the day