KiCad/kicad-symbols #2971: LM358, LM2902

g200kg, 2020-09-05:
I'm not sure if it's a library problem, 
Amplifire_Operational  LM358 Unit B,  LM2902 Unit B,C,D shows unconnected pin mark indicating different from the actual connection.

[image]

chschlue, 2020-09-05:
[image]

Can't reproduce the issue.
Please add more information (like KiCad version, exact steps leading to your issue.)

(I doubt it's a library problem, there is no difference between the pins that behave differently for you.)

g200kg, 2020-09-05:
Thank you for the reply.

KiCad 5.1.6 on ubuntu 20.04.1 and KiCad 5.1.5 on win10
Both have the same result.

I found the procedure of reproduction.

1. Place LM358 "Unit A" from Amplifier_Operational.
2. Change the placed "Unit A" to "Unit B" by selecting [Properties]-[Unit] from the right-click menu.
3. connect pin6 and pin 7. 

This issue does not occur if "Unit B" is selected during initial deployment
This may be an issue with the app, not the library?

------
Application: KiCad
Version: 5.1.6-c6e7f7d~87~ubuntu20.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-45-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

------
Application: KiCad
Version: (5.1.5)-3, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

chschlue, 2020-09-05:
> This may be an issue with the app, not the library?

Almost certainly.
Please open an issue on https://gitlab.com/kicad/code/kicad/-/issues

Closing for now.

closed 2020-09-05 by chschlue