KiCad/kicad-symbols #2984: [v6] CI

chschlue, 2020-09-19:


cpresser, 2020-09-19:
I already started on this one.
My WIP stuff is here:
* https://gitlab.com/cpresser/kicad-footprints
* https://gitlab.com/cpresser/kicad-library-utils

To test it, I also made 3 merge requests:
https://gitlab.com/cpresser/kicad-footprints/-/merge_requests
* One with errors
* One without errors
* One that does not change a footprint

So far that looks good.

cpresser, 2020-09-27:
I have started looking into symbols. Yesterday was the first time I ran kicad6.

The documentation (https://docs.google.com/document/d/1AiPZvWT_QBPyVQSyn3ISl6jVntjrrtATEYxzBFJM1HU/edit) has a disclaimer about being outdated, and yes, that seems to be true. For example I can't find the word `extends` in the docu. But fortunately the format seems to be mostly self explanatory.

I think I can start from `pcb/kicad_mod.py` since it already has most of the sexpr parsing and convenience functions. Has anyone else already put work into making a python parser for the `.kicad_sym` files? I want to prevent doing duplicate work here.