KiCad/kicad-symbols #3061: [v6] Memory_*

chschlue, 2020-09-19:


cpresser, 2020-09-19:
Is there a use-case where people actually connect a SPI flash in daisy-chain configuration?
I am not aware of one, so I think it is a good idea to put all ports on the same side.
**Perhaps like so**
[image]

----

Here are some of the current AT25 style flashes. All look different:
[image]
[image]
[image]
[image]
[image]

chschlue, 2020-09-20:
Your example looks good to me.
Daisy chaining in conjunction with large reads doesn't make sense to me.